espressif/esp-serial-flasher #129: ESP32-cam flashing not working

johnyHV, 2025-01-02:
### Port

ESP32

### Target chip

ESP32

### Hardware Configuration

Ai Thinker ESP32-cam with MB board. GND/R revision
[image]

### Log output

```plain
esptool.js
Serial port WebSerial VendorID 0x1a86 ProductID 0x7523
Connecting...Debug: using Classic Serial Reset
Debug: _connect_attempt default_reset
Debug: Sync connect attempt 0
Debug: Sync
Debug: Sync err Error: Invalid head of packet (0x74): Possible serial noise or corruption.
Debug: Error at sync Error: Invalid head of packet (0x74): Possible serial noise or corruption.
Debug: Sync connect attempt 1
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 2
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 3
Debug: Sync
Debug: 538052359
Debug: Connect attempt successful.

Debug: Chip Magic f01d83
Detecting chip type... ESP32
Debug: Read efuse 1073061900
Debug: Read efuse 1073061900
Debug: Read efuse 1073061908
Debug: Read efuse 1073061900
Chip is ESP32-D0WDQ6-V3 (revision 3)
Debug: Read efuse 1073061900
Debug: Read efuse 1073061900
Debug: Read efuse 1073061904
Debug: Read efuse 1073061912
Features: Wi-Fi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
Debug: Read efuse 1073061892
Debug: Read efuse 1073061896
MAC: 08:3a:f2:67:24:94
Debug: Read efuse 1073061892
Debug: Read efuse 1073061896
Uploading stub...
Debug: mem_begin 3532 1 6144 400be000
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Debug: mem_begin 156 1 6144 3ffdeba8
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Running stub...
Debug: check_command leave RAM download mode
Stub running...
Changing baudrate to 460800
Changed
Debug: EspLoader program
Debug: Data Length 1601968
Debug: Image Length 1601968
Debug: _update_image_flash_params keep undefined undefined
Debug: Image MD5 761fe8137e96e56708ee722eda152695
Compressed 1601968 bytes to 932891...
Debug: check_command enter compressed flash mode
Error: No serial data received.
```

### More Information

FW is generate from arduino IDE. Flashing FW from Arduino IDE or oficial windows application ESP flash download tool working fine. But flashing FW from WEB app don't work. 

Here is log from Arduino IDE.

```"C:\Users\Miroslav\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.9.dev3/esptool.exe" --chip esp32 --port "COM14" --baud 460800  --before default_reset --after hard_reset write_flash  -z --flash_mode keep --flash_freq keep --flash_size keep 0x1000 "C:\Users\Miroslav\AppData\Local\arduino\sketches\CD79AF74A217813F66BE2EC977F0AE21/ESP32_PrusaConnectCam.ino.bootloader.bin" 0x8000 "C:\Users\Miroslav\AppData\Local\arduino\sketches\CD79AF74A217813F66BE2EC977F0AE21/ESP32_PrusaConnectCam.ino.partitions.bin" 0xe000 "C:\Users\Miroslav\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0/tools/partitions/boot_app0.bin" 0x10000 "C:\Users\Miroslav\AppData\Local\arduino\sketches\CD79AF74A217813F66BE2EC977F0AE21/ESP32_PrusaConnectCam.ino.bin" 
esptool.py v4.8.1
Serial port COM14
Connecting.......
Chip is ESP32-D0WDQ6-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:3a:f2:67:24:94
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00197fff...
Compressed 24976 bytes to 15953...
Writing at 0x00001000... (100 %)
Wrote 24976 bytes (15953 compressed) at 0x00001000 in 0.9 seconds (effective 227.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 307.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 435.5 kbit/s)...
Hash of data verified.
Compressed 1602128 bytes to 932972...
Writing at 0x00010000... (1 %)
Writing at 0x0001b776... (3 %)
Writing at 0x0002a439... (5 %)
Writing at 0x00035464... (7 %)
Writing at 0x0004109c... (8 %)
Writing at 0x00050f01... (10 %)
Writing at 0x00059e49... (12 %)
Writing at 0x0006040f... (14 %)
Writing at 0x000685fb... (15 %)
Writing at 0x0007b10b... (17 %)
Writing at 0x00083799... (19 %)
Writing at 0x0008c024... (21 %)
Writing at 0x000949cc... (22 %)
Writing at 0x0009bada... (24 %)
Writing at 0x000a1cba... (26 %)
Writing at 0x000a83d9... (28 %)
Writing at 0x000ae4b5... (29 %)
Writing at 0x000b4326... (31 %)
Writing at 0x000b9df9... (33 %)
Writing at 0x000bf810... (35 %)
Writing at 0x000c53e1... (36 %)
Writing at 0x000cad51... (38 %)
Writing at 0x000cffda... (40 %)
Writing at 0x000d574c... (42 %)
Writing at 0x000dad9c... (43 %)
Writing at 0x000e0456... (45 %)
Writing at 0x000e5dfc... (47 %)
Writing at 0x000eb1d4... (49 %)
Writing at 0x000f05cb... (50 %)
Writing at 0x000f5dca... (52 %)
Writing at 0x000fb073... (54 %)
Writing at 0x001020b7... (56 %)
Writing at 0x001079bb... (57 %)
Writing at 0x0010cf81... (59 %)
Writing at 0x0011267a... (61 %)
Writing at 0x00117825... (63 %)
Writing at 0x0011cfbd... (64 %)
Writing at 0x00122d85... (66 %)
Writing at 0x0012867b... (68 %)
Writing at 0x0012dbbc... (70 %)
Writing at 0x00132efc... (71 %)
Writing at 0x001381ab... (73 %)
Writing at 0x0013d822... (75 %)
Writing at 0x00142ff4... (77 %)
Writing at 0x001483c2... (78 %)
Writing at 0x0014d7ec... (80 %)
Writing at 0x0015300c... (82 %)
Writing at 0x00158f55... (84 %)
Writing at 0x0015ea6d... (85 %)
Writing at 0x00165ad8... (87 %)
Writing at 0x001702f3... (89 %)
Writing at 0x001762b2... (91 %)
Writing at 0x0017b3f5... (92 %)
Writing at 0x00180bc8... (94 %)
Writing at 0x00186800... (96 %)
Writing at 0x0018bd90... (98 %)
Writing at 0x00191561... (100 %)
Wrote 1602128 bytes (932972 compressed) at 0x00010000 in 22.7 seconds (effective 564.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...```

Dzarda7, 2025-01-02:
Hi @johnyHV, I think you accidentally opened the issue in the wrong repo. This one is for flashing Espressif SoCs from another microcontroller. I think you you probably open the issue in [esptool-js](https://github.com/espressif/esptool-js) repo.

johnyHV, 2025-01-03:
Hi @Dzarda7 . Oh, yes. sorry. My mistake.

closed 2025-01-03 by johnyHV

Dzarda7, 2025-01-03:
FYI: you might find this project useful as I saw that Prusa buddy firmware already uses this to flash wifi modules. Same can be done for the esp camera.