espressif/esp-serial-flasher #130: cant compile esp32 example

mtraven, 2025-01-11:
trying to use a ESP32 WROOM devboard1 to flash an esp32 that doesn't have a usb interface.  Getting hung up trying to compile the esp32 example, which seems like exactly what I need.  I'm calling idf.py build (sometimes reconfig) from \examples\esp32_example.  I can't get it to get past the configuration. I also cant get into menuconfig.  I've tried this with at least 3 different frameworks, most recently 5.1.2 that you see there.  I'm doing this all in vscode if that matters. I'm sure you'll need more information, but I dont know what that will be, so ask and I'll do my best to provide it.  Thanks!

here is the the diagnostic that comes out when the compilation fails:

`CMake Error at C:/esp/frameworks/v5.1.2/esp-idf/tools/cmake/component.cmake:381 (__component_get_property):  
  __component_get_property Function invoked with incorrect arguments for
  function named: __component_get_property
Call Stack (most recent call first):
  CMakeLists.txt:11 (idf_component_get_property)
`

mtraven, 2025-01-12:
The first framework installed was the standalone IDE + framework from esspressIf.  I dont use that IDE, but it does compile example projects just fine.  The remaining frameworks were installed via the IDF plugin in VScode.  Those frameworks all compile example files properly as well.

I have tried several different things to put the project into VScode, though I suspect thats where I'm making some mistake.
The first way was to open the whole repo folder with vs code, save the workspace and then use an IDF terminal to navigate to the example folder and run idf.py build.
The second method I tried was to open just the example folder with VScode and run idf.py build from there (via the button or CL)
the third method was to try an not use VS code at all and just to it from a CL (that standalone IDE I installed gives me a shortcut to start that python environment.

All of these methods fail during the configuration part of setup, something about getting component properties that I dont really understand.

Thanks so much for taking the time to help, though I write a but of code, I'm not a coder...so sometimes even the basic stuff confuses me.

**as a side note;  since you asked about vscode working with regular examples, I should mention that I have had some recent trouble with this.  I hadn't used esspressIF for about 6 months and when I went back into it, none of my projects would compile.  I spent like 2 days trying to fix that, and I think I have....it compiles other stuff now, just not this repo.

mtraven, 2025-01-12:
Update:  was checking an old project that used 4.4.7, thought I'd take a shot and try it with the esp32-esp32 example....to my shock (i woke people up in my house) it worked! at least the example did, now I gotta figure out how to wire & use it, but thats a big win.

I must have missed an IDF version requirement....what version had you done the test compile on?  What versions should it work with?

Dzarda7, 2025-01-12:
The supported versions of ESP-IDF is anything higher than v4.3 as stated in the main [README](https://github.com/espressif/esp-serial-flasher?tab=readme-ov-file#supported-esp-idf-versions). We are testing all the stable releases since v4.3 in the CI and the build of this example passes. There seems to be something wrong with your environment, I would suggest you to use the latest ESP-IDF which is v5.4 and follow the [documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/index.html) properly.

Now regarding the use of custom binaries. The binaries we use in the example are in [binaries directory](https://github.com/espressif/esp-serial-flasher/tree/master/examples/binaries). All of these are bundled with the example as there is a script in [CMakeLists.txt](https://github.com/espressif/esp-serial-flasher/blob/c7b8433146631a482c1f0b5d023f89f25f517354/examples/esp32_example/main/CMakeLists.txt#L12) of each example that creates a binary.c file which contains binaries from the directory converted to C array (the binaries.c is in the build directory of the project). You do not really need to use it like this if you are making your own project. You can include esp-serial-flasher as external component from [Espressif Component Registry](https://components.espressif.com/components/espressif/esp-serial-flasher/versions/1.7.0) and use `esp_loader.h` api with reuse of some [example_common.c](https://github.com/espressif/esp-serial-flasher/blob/master/examples/common/example_common.c) functions. This way you can even flash some binary files from sd card for example. There is [link](https://github.com/Dzarda7/esf-demo) for the demo I made the way I just described, but it is a bit more complex. If you just want to use the `esp32_example` with your binaries, you can just replace the ones in the [binaries directory](https://github.com/espressif/esp-serial-flasher/tree/master/examples/binaries/Hello-world) with the same names.

Hope this helps.

mtraven, 2025-01-12:
thanks for the response.  Not sure why you think something is wrong with my environment...I've tested this on 5.4, 5.3.2 5.12 and 4.4.7...447 is the only one that compiles ...the rest of the frameworks compile examples and my personal projects.  I even installed a fresh version of windows on a drive to start over, same error with all of them.

But I did manage to flash my esp32 by swapping out the bin files in hello world.  Thank you for the instructions & example on how to do that properly, I'll work my way through that tonight.

Thanks so much for your help, I was three days in and ready to give up when I posted on here...now I'm close to having the whole project working.  All in service of adding wifi to my 3d printer.

Dzarda7, 2025-01-13:
Because we build it with all these versions and it works. I even tried to install new 5.4 release on windows machine and it still works, so sorry, but I am not able to reproduce your issue unfortunately. I would try [manual installation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/index.html#manual-installation) and just build it using `idf.py build` in `examples/esp32_example` directory. But if you are ok with 4.4.7, feel free to use that.

dobairoland, 2025-01-13:
Please open an issue in https://github.com/espressif/vscode-esp-idf-extension and explain how did you set up those newest ESP-IDF environments. Probably they can help you more and even ensure that such issue won't happen again.

I'm closing this because it is not esp-serial-flasher issue.

closed 2025-01-13 by dobairoland

IntegratedTurnover, 2025-02-12:
I'm having the same problem as [mtraven](https://github.com/mtraven), my console output looks much the same:
```
C:\Users\jharris\Desktop\Trash\2108Updater\examples\esp32_example>idf.py build
Executing action: all (aliases: build)
Running cmake in directory C:\Users\jharris\Desktop\Trash\2108Updater\examples\esp32_example\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif-jake\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\Users\jharris\Desktop\Trash\2108Updater\examples\esp32_example"...
-- IDF_TARGET is not set, guessed 'esp32' from sdkconfig 'C:/Users/jharris/Desktop/Trash/2108Updater/examples/esp32_example/sdkconfig'
-- Found Git: C:/Espressif-jake/tools/idf-git/2.44.0/cmd/git.exe (found version "2.44.0.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif-jake/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif-jake/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif-jake/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
NOTICE: Skipping optional dependency: espressif/usb_host_cdc_acm
Processing 1 dependencies:
[1/1] idf (5.2.2)
NOTICE: Skipping optional dependency: espressif/usb_host_cdc_acm
-- Project sdkconfig file C:/Users/jharris/Desktop/Trash/2108Updater/examples/esp32_example/sdkconfig
Loading defaults file C:/Users/jharris/Desktop/Trash/2108Updater/examples/esp32_example/sdkconfig.defaults...
-- Compiler supported targets: xtensa-esp-elf
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: C:/Espressif-jake/python_env/idf5.2_py3.11_env/Scripts/python.exe (found version "3.11.2") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "esp-serial-flasher" version: 1
-- Adding linker script C:/Users/jharris/Desktop/Trash/2108Updater/examples/esp32_example/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/jharris/Desktop/Trash/2108Updater/examples/esp32_example/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: 2108Updater app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/jharris/Desktop/Trash/2108Updater C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/app_trace C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/app_update C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/bootloader C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/bootloader_support C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/bt C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/cmock C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/console C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/cxx C:/Espressif-jake/frameworks/esp-idf-v5.2.2/components/driver C:/Espressif-jake/frameworks/esp-idf-v5 [... cut ...]
This isn't a vscode or eclipse issue, since I have neither installed on this pc. I've manually installed esp-idf from the get-go and have never installed vscode or eclipse since the esp toolchain is a house of cards all on its own. This setup has compiled other projects without issue, so much like mtraven I know my setup works.

Since I'm getting the same complaints about CONFIG_SOC_BROWNOUT_RESET_SUPPORTED only with this project, I tend to believe it's something to do with this project.

At any rate, deleting lines 10 through 18 (the if statement pertaining to the IDF version) of examples\esp32_example\CMakeLists.txt allows the project to compile and everything seems hunky-dory.

Dzarda7, 2025-02-13:
@IntegratedTurnover @mtraven sorry guys for incorrect conclusion. The problem is that ESP-IDF by default uses directory name as component name. That is why it fails when you rename esp-serial-flasher directory, because it is checked against `esp-serial-flasher` name using `idf_component_get_property` in the section @IntegratedTurnover mentioned. I suppose that this might be the problem. I tested it now and it is the problem. Solution is to build it in the original directory or change the `esp-serial-flasher` to your directory name in [this line](https://github.com/espressif/esp-serial-flasher/blob/7a109ebf22088ab5d91470a36edd71885cbec0d7/examples/esp32_example/CMakeLists.txt#L11).

I noticed it now and I think it is not ideal to have it like this, so I will think of a better solution.