espressif/esp-serial-flasher #131: timeout_per_mb does not scale if size < 1e6

haydenroche5, 2025-01-16:
### Port

ESP

### Target chip

ESP32S3

### Hardware Configuration

N/A

### Log output

```plain
N/A
```

### More Information

Context: https://discuss.blues.com/t/outboard-dfu-failure-with-larger-binaries/1947/9?u=hroche

Proposed fix:

```
static uint32_t timeout_per_mb(uint32_t size_bytes, uint32_t time_per_mb)
{
    uint32_t timeout = ((uint64_t) time_per_mb * size_bytes) / 1000000;
    return MAX(timeout, DEFAULT_FLASH_TIMEOUT);
}
```

Dzarda7, 2025-01-16:
Hi @haydenroche5, thanks for noticing this. I will fix that. You also need a new release of esp-serial-flasher or you okay with master?

haydenroche5, 2025-01-16:
Master's good. Thank you!

Dzarda7, 2025-01-16:
I took a closer look now and I cannot agree with you, if I did not miss something. When `size_bytes` is divided by 1e6 (which is float), it is cast to float, then multiplied by `time_per_mb` (output is also float) and after that converted to `uint32_t`. You can try it [here](https://www.programiz.com/online-compiler/7vCy2oE6baDQ8), where I prepared the exact example.

haydenroche5, 2025-01-16:
Ah, you are right. I had changed 1e6 to 1000000 and that of course causes the problem I was seeing. Sorry for the false alarm.

Dzarda7, 2025-01-16:
No problem, if you find the bug somewhere else, feel free to open the issue.

closed 2025-01-16 by Dzarda7