espressif/esp-serial-flasher #132: How to flash on host?

KammutierSpule, 2025-03-11:
### Is your feature request related to a problem?

Is possible using this library to compile and run it on host? (as an alternative to esptool.py)?
Any of these examples would help ?

### Describe the solution you'd like

Be able to use this library to compile an application on PC

dobairoland, 2025-03-12:
Hi @KammutierSpule. This request is tracked in https://github.com/espressif/esp-serial-flasher/pull/102. It is not supported in this moment and has a low priority for us now.

KammutierSpule, 2025-03-17:
@Dzarda7 Linux would be best option for me. I was wondering if this library should be implemented generic without target specific. You would just have to implement the target (UART and GPIO) functions.

However, currently, I'm using a ESP which I program via USB (/dev/ttyACM0) with esptool.py
I don't know if this tool is using same uart protocol or there is some specific way of programming this ESPs when using USB port.

Dzarda7, 2025-03-17:
Yes, it is supposed to work that way so you can implement whatever port you want. There are just some already done, like for example Raspberry pi port. You can definitely add another port you want according to the [docs](https://github.com/espressif/esp-serial-flasher?tab=readme-ov-file#supporting-a-new-host-target). It should be simple enough I believe. My point was that I will try to rewrite rpi port to use some kernel modules to interact with UART (and with USB maybe if it will be possible), so it will not be hardware specific.

BTW this tool uses same protocol for communication with chips as esptool, it is described [here](https://docs.espressif.com/projects/esptool/en/latest/esp32h2/advanced-topics/serial-protocol.html).

Dzarda7, 2025-03-24:
I will close this issue as we already have a tracker for this. Thanks for pointing this out, I will try to implement the linux support but cannot promise anything and do not know the timeframe. For other systems I am not sure if it makes sense as there is already espool.

KammutierSpule, 2025-03-26:
@Dzarda7  I has a look on the library and I understand eg: the Raspbery port, it uses RX, TX and Reset and GPIO0 pins. however, for ESP devices with USB, they only have eg /dev/ttyACM0, how is it is supposed to use the virtual USB serial port created by the ESP to connect to bootloader?

Dzarda7, 2025-03-28:
@KammutierSpule you can get the chips into bootloader using toggling RTS and DTR signals through the USB. There are two options for flashing the device through USB, either the chip can be connected through USB-serial-JTAG or through USB-UART converted (for example CP2102), the approach of toggling the RTS and DTR works in both cases, just the sequence is a bit different I believe. This approach is used in [esp32_usb_cdc_acm_port.c](https://github.com/espressif/esp-serial-flasher/blob/bebb8d293c7fe4c60bb8053dede2b1aaa383ff80/port/esp32_usb_cdc_acm_port.c#L94). If you are more interested in these sequences, take a look at esptool documentation or code.

As I said, you can use whatever hardware layer you want with this library, it does not care through what are the data transferred. You just need to create the port in the [port folder](https://github.com/espressif/esp-serial-flasher/tree/master/port) and in add it into main [CMakeLists.txt](https://github.com/espressif/esp-serial-flasher/blob/master/CMakeLists.txt) as stated in the main [README.md](https://github.com/espressif/esp-serial-flasher/tree/master?tab=readme-ov-file#supporting-a-new-host-target).

I did a bit of a research and tried a few things and I believe that termios library is ideal for this case for linux. Not sure if you can toggle RTS and DTR through it or you need something else, but this should work. If you are willing to give it a try and implement it, I would be really happy to help you and merge it into this repo. I really see the benefit here as you can run it on some less powerful SBC without python for example.