espressif/esp-serial-flasher #133: "lower" addresses work but "higher" do not

Fabchick, 2025-04-08:
### Port

ESP, UART

### Target chip

ESP32-C6

### Hardware Configuration

I am using a custom PCB. I am using a ESP32-S3-WROOM-1 and a ESP32-C6-WROOM-1. The S3 is flashing the C6.
They are connected by UART TX/RX with each other. ESP32-S3 has one GPIO connected with the Reset Pin of  ESP32-C6 and one GPIO connected with the Boot0 Pin of ESP32-C6.
Baudrate is configured at 115200.
See espLoaderConfig:
```
  const loader_esp32_config_t espLoaderConfig = {
      .baud_rate = 115200, 
      .uart_port = UART_NUM_0,
      .uart_rx_pin = GPIO_NUM_44,
      .uart_tx_pin = GPIO_NUM_43,
      .reset_trigger_pin = GPIO_NUM_6,
      .gpio0_trigger_pin = GPIO_NUM_5,
      .dont_initialize_peripheral = true,
  };
```

### Log output

```plain
At first the "good" output, when the address is set to 0x20000:
Connecting...
esp_loader_connect_secure_download_mode()

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 
--- READ ---
49 20 28 31 30 34 29 20 62 6f 6f 74 3a 20 20 32 20 70 68 79 5f 69 6e 69 74 20 20 20 20 20 20 20 20 20 52 46 20 64 61 74 61 20 20 20 20 20 20 20 20 20 20 30 31 20 30 31 20 30 30 30 31 62 30 30 30 20 30 30 30 30 31 30 30 30 1b 5b 30 6d 0d 0a 1b 5b 30 3b 33 32 6d 49 20 28 31 31 31 29 20 62 6f 6f 74 3a 20 20 33 20 66 61 63 74 6f 72 79 20 20 20 20 20 20 20 20 20 20 66 61 63 74 6f 72 79 20 61 70 70 20 20 20 20 20 20 30 30 20 30 30 20 30 30 30 32 30 30 30 30 20 30 30 32 30 30 30 30 30 1b 5b 30 6d 0d 0a 1b 5b 30 3b 33 32 6d 49 20 28 31 31 39 29 20 62 6f 6f 74 3a 20 20 34 20 6f 74 61 5f 30 20 20 20 20 20 20 20 20 20 20 20 20 4f 54 41 20 61 70 70 20 20 20 20 20 20 20 20 20 20 30 30 20 31 30 20 30 30 32 32 30 30 30 30 20 30 30 32 30 30 30 30 30 1b 5b 30 6d 0d 0a 1b 5b 30 3b 33 32 6d 49 20 28 31 32 37 29 20 62 6f 6f 74 3a 20 20 35 20 6f 74 61 5f 31 20 20 20 20 20 20 20 20 20 20 20 20 4f 54 41 20 61 70 70 20 20 20 20 20 20 20 20 20 20 30 30 20 31 31 20 30 30 34 32 30 30 30 30 20 30 30 32 30 30 30 30 30 1b 5b 30 6d 0d 0a 1b 5b 30 3b 33 32 6d 49 20 28 31 33 34 29 20 62 6f 6f 74 3a 20 20 36 20 73 74 6f 72 61 67 65 20 20 20 20 20 20 20 20 20 20 55 6e 6b 6e 6f 77 6e 20 64 61 74 61 20 20 20 20 20 30 31 20 38 31 20 30 30 36 32 30 30 30 30 20 30 30 31 30 30 30 30 30 1b 5b 30 6d 0d 0a 1b 5b 30 3b 33 32 6d 49 20 28 31 34 32 29 20 62 6f 6f 74 3a 20 20 37 20 6e 76 73 5f 6b 65 79 20 20 20 20 20 20 20 20 20 20 4e 56 53 20 6b 65 79 73 20 20 20 20 20 20 20 20 20 30 31 20 30 34 20 30 30 37 32 30 30 30 30 20 30 30 30 30 31 30 30 30 1b 5b 30 6d 0d 0a 1b 5b 30 3b 33 32 6d 49 20 28 31 34 39 29 20 62 6f 6f 74 3a 20 20 38 20 63 6f 72 65 64 75 6d 70 20 20 20 20 20 20 20 20 20 55 6e 6b 6e 6f 77 6e 20 64 61 74 61 20 20 20 20 20 30 31 20 30 33 20 30 30 37 32 31 30 30 30 20 30 30 30 31 30 30 30 30 1b 5b 30 6d 6d 6f 64 65 29 0d 0a 
--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 
--- READ ---
c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 
--- WRITE ---
c0 00 0d 08 00 00 00 00 00 00 00 00 00 00 00 00 00 c0 
--- READ ---
c0 01 0d 04 00 00 00 00 00 00 00 00 00 c0 Connected to target
Flashing ota binary(file)
Flashing 790528 bytes to address 0x00020000
Erasing flash (this may take a while)...

--- WRITE ---
c0 00 02 14 00 00 00 00 00 00 10 0c 00 04 03 00 00 00 04 00 00 00 00 02 00 00 00 00 00 c0 
--- READ ---
c0 01 02 04 00 00 00 00 00 00 00 00 00 c0 Flash erased

--- WRITE ---
c0 00 03 10 04 db dd 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 05 be 82 2e 5e f9 2d 30 50 e7 98 e4 b5 50 bb 3b 10 29 34 55 a7 2b 61 c3 66 d5 75 68 2e f7 88 9b c3 64 d0 97 c1 ac b5 d6 17 0a 6e 57 3b 7f 4e 68 ee 77 3c 71 3d 14 24 62 ab 3d 9f 5f 52 87 ff 59 9d ec 9c 8f 98 7e f0 72 8d 47 5c f3 16 35 28 2b a3 74 ad cd 1c bc a1 d8 c1 1b 21 53 cf fd de f1 2f a2 d4 c1 4d db dd a0 e0 92 fd 46 40 d1 86 a1 b6 26 8a 0b 41 c3 62 38 9f 9f db dc 98 07 83 e7 3e 8e 57 82 ee f8 f8 80 e7 22 17 48 30 a7 db dd b3 70 21 84 ec 97 b0 74 9a 98 85 41 8e 38 15 d6 4b 53 05 80 d7 07 a7 f6 b4 53 a5 f9 9e c8 cb 95 8d ba b3 f9 32 c1 69 fb 04 ec ac 44 15 2a 30 db dd b6 55 bd 05 b2 8f 37 78 10 fd 8f 20 87 e7 21 e2 9a 6a 15 45 84 8f a5 da 0e fd 99 f5 b2 10 29 46 46 f8 de 28 45 93 5f 60 fc a3 29 cd b7 d2 8e 4a 2a de 28 13 aa 1b d1 4c 97 1b 31 e5 b3 58 36 0f fc a4 19 30 77 4f c6 2c fd c3 9b 38 7a 42 42 5b a9 91 40 9f 0d 9b 03 ab 13 ad 21 55 ca 6b 48 69 5f 51 f5 76 a8 ea 91 2d 21 1c 0b e6 c7 8c a2 a1 3f 0b d1 60 ce 94 [...] // this works and will go on

Now the "bad" output, when the address is set to 0x220000:
Connecting...
esp_loader_connect_secure_download_mode()

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 
--- READ ---
30 30 32 30 30 30 30 30 1b 5b 30 6d 0d 0a 1b 5b 30 3b 33 32 6d 49 20 28 31 32 37 29 20 62 6f 6f 74 3a 20 20 35 20 6f 74 61 5f 31 20 20 20 20 20 20 20 20 20 20 20 20 4f 54 41 20 61 70 70 20 20 20 20 20 20 20 20 20 20 30 30 20 31 31 20 30 30 34 32 30 30 30 30 20 30 30 32 30 30 30 30 30 1b 5b 30 6d 0d 0a 1b 5b 30 3b 33 32 6d 49 20 28 31 33 34 29 20 62 6f 6f 74 3a 20 20 36 20 73 74 6f 72 61 67 65 20 20 20 20 20 20 20 20 20 20 55 6e 6b 6e 6f 77 6e 20 64 61 74 61 20 20 20 20 20 30 31 20 38 31 20 30 30 36 32 30 30 30 30 20 30 30 31 30 30 30 30 30 1b 5b 30 6d 0d 0a 1b 5b 30 3b 33 32 6d 49 20 28 31 34 32 29 20 62 6f 6f 74 3a 20 20 37 20 6e 76 73 5f 6b 65 79 20 20 20 20 20 20 20 20 20 20 4e 56 53 20 6b 65 79 73 20 20 20 20 20 20 20 20 20 30 31 20 30 34 20 30 30 37 32 30 30 30 30 20 30 30 30 30 31 30 30 30 1b 5b 30 6d 0d 0a 1b 5b 30 3b 33 32 6d 49 20 28 31 34 39 29 20 62 6f 6f 74 3a 20 20 38 20 63 6f 72 65 64 75 6d 70 20 20 20 20 20 20 20 20 20 55 6e 6b 6e 6f 77 6e 20 64 61 74 61 20 20 20 20 20 30 31 20 30 33 20 30 30 37 32 31 30 30 30 20 30 30 30 31 30 30 30 30 1b 5b 30 6d 0d 0a 1b 5b 30 3b 33 32 6d 49 20 28 31 35 37 29 20 62 6f 6f 74 3a 20 45 6e 64 20 6f 66 20 70 61 72 74 69 74 69 6f 6e 20 74 61 62 6c 65 1b 5b 30 6d 0d 0a 1b 5b 30 3b 33 32 6d 49 20 28 31 36 31 29 20 62 6f 6f 74 3a 20 44 65 66 61 75 6c 74 69 6e 67 20 74 6f 20 66 61 63 74 6f 72 79 20 69 6d 61 67 65 1b 5b 30 6d 0d 0a 1b 5b 30 3b 33 32 6d 49 20 28 31 36 36 29 20 65 73 70 5f 69 6d 61 67 65 3a 20 73 65 67 6d 65 6e 74 20 30 3a 20 70 61 64 64 72 3d 30 30 30 32 30 30 32 30 20 76 61 64 64 72 3d 34 32 30 38 30 30 32 30 20 73 69 7a 65 3d 31 64 32 36 30 68 20 28 31 31 39 33 39 32 29 20 6d 61 70 1b 5b 30 6d 0d 0a 62 6f 6f 74 3a 30 78 32 34 20 28 44 4f 57 4e 4c 4f 41 44 28 55 53 42 2f 55 41 52 54 30 2f 53 44 49 4f 5f 46 45 49 5f 46 45 4f 29 29 0d 0a 77 61 69 74 20 75 61 72 74 20 64 6f 77 6e 6c 6f 61 64 28 73 65 63 75 72 65 20 6d 6f 64 65 29 0d 0a 
--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 
--- READ ---
c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 
--- WRITE ---
[... 50 lines cut ...]
# CONFIG_SERIAL_FLASHER_RESET_INVERT is not set
# CONFIG_SERIAL_FLASHER_BOOT_INVERT is not set
# end of ESP serial flasher
```

I can also successfully flash the otadata-partition located at 0xd000.

With the esptool.py I can also flash to the address 0x220000 without any issues.

Something must be wrong in this sequence:
--- WRITE ---
c0 00 02 14 00 00 00 00 00 00 10 0c 00 04 03 00 00 00 04 00 00 00 00 22 00 00 00 00 00 c0 
--- READ ---
c0 01 02 04 00 00 00 00 00 01 06 00 00 c0 DEBUG: Error: COMMAND_FAILED

But I do not know how to fix it.
On address 0x20000 it look like this:
--- WRITE ---
c0 00 02 14 00 00 00 00 00 00 10 0c 00 04 03 00 00 00 04 00 00 00 00 02 00 00 00 00 00 c0 
--- READ ---
c0 01 02 04 00 00 00 00 00 00 00 00 00 c0 Flash erased

What is the 01 06 in the --- READ --- meaning? Why is 22 00 00 00 00 00 c0 not working, but 02 00 00 00 00 00 c0 is working?

Is there an issue with a maximum Flash size configuration?

Dzarda7, 2025-04-09:
Hi @Fabchick, thanks for the report of this issue. I am not aware of any max flash size configuration. Regarding the message you see, the flashing uses Serial protocol and the `01 06` are the status bytes at the end of the data as described [here](https://docs.espressif.com/projects/esptool/en/latest/esp32c6/advanced-topics/serial-protocol.html#status-bytes). This exact sequence for example means that ROM bootloader failed to act on the command. I will try to investigate it a bit more. For now it seems that it might fail after 0x200000 address, can you please verify that? Firstly write to it and the please try `0x200000 - data size + 1`. Thanks.

Fabchick, 2025-04-09:
Hi @Dzarda7, thanks for the quick response and the further information.
I have tried your suggestion and some other addresses:
❌ = error, ✅ = success

❌ 0x200000 - data size + 1
Flashing 790528 bytes to address **0x0013F001**
Erasing flash failed with error: 5
If using Secure Download Mode, double check that the specified                    target flash size is correct.

✅ 0x200000 - data size
Flashing 790528 bytes to address **0x0013F000**
--- WRITE ---
c0 00 02 14 00 00 00 00 00 00 10 0c 00 04 03 00 00 00 04 00 00 00 f0 13 00 00 00 00 00 c0 
--- READ ---
c0 01 02 04 00 00 00 00 00 00 00 00 00 c0 Flash erased

--- WRITE ---
c0 00 03 10 04 db dd 00 [...]

❌ 0x200000 - data size + 0x100
Flashing 790528 bytes to address **0x0013F100**
--- WRITE ---
c0 00 02 14 00 00 00 00 00 00 10 0c 00 04 03 00 00 00 04 00 00 00 f1 13 00 00 00 00 00 c0 
--- READ ---
c0 01 02 04 00 00 00 00 00 01 06 00 00 c0 DEBUG: Error: COMMAND_FAILED
Erasing flash failed with error: 9

❌ 0x200000 - data size - 0x100
Flashing 790528 bytes to address **0x0013EF00**
--- WRITE ---
c0 00 02 14 00 00 00 00 00 00 10 0c 00 04 03 00 00 00 04 00 00 00 ef 13 00 00 00 00 00 c0 
--- READ ---
c0 01 02 04 00 00 00 00 00 01 06 00 00 c0 DEBUG: Error: COMMAND_FAILED
Erasing flash failed with error: 9

❌ 0x200000 - data size + 0x1000
flashing 790528 bytes to address **0x00140000**
--- WRITE ---
c0 00 02 14 00 00 00 00 00 00 10 0c 00 04 03 00 00 00 04 00 00 00 00 14 00 00 00 00 00 c0 
--- READ ---
c0 01 02 04 00 00 00 00 00 01 06 00 00 c0 DEBUG: Error: COMMAND_FAILED
Erasing flash failed with error: 9

✅ 0x200000 - data size - 0x1000
Flashing 790528 bytes to address **0x0013E000**
--- WRITE ---
c0 00 02 14 00 00 00 00 00 00 10 0c 00 04 03 00 00 00 04 00 00 00 e0 13 00 00 00 00 00 c0 
--- READ ---
c0 01 02 04 00 00 00 00 00 00 00 00 00 c0 Flash erased

--- WRITE ---
c0 00 03 10 04 db dd 00 00 00 00 04 [...]

some more:
❌ 0x200000
❌ 0x1F0000
✅ 0x100000
❌ 0x1A0000
❌ 0x160000
✅ 0x120000

It seems that there is definitely a barrier at **0x200000**.
Tomorrow I want to test this with a setup without secure boot and flash encryption

Fabchick, 2025-04-10:
I just tested it without flash encryption and secure boot:
When I use the function `esp_loader_connect_secure_download_mode()` - I get the same error.
```
  esp_loader_connect_args_t connect_config = ESP_LOADER_CONNECT_DEFAULT();
  esp_loader_error_t err = esp_loader_connect_secure_download_mode(&connect_config,
                                                                   8 * 1024 * 1024,
                                                                   ESP32C6_CHIP);
```

But when I use the function `esp_loader_connect()` - It works! 
```
  esp_loader_connect_args_t connect_config = ESP_LOADER_CONNECT_DEFAULT();
  esp_loader_error_t err = esp_loader_connect(&connect_config);
```

But I have to enable flash encryption and secure boot. Using `esp_loader_connect()` gives me the same error with these modes enabled. Any suggestions what I could test next?

Dzarda7, 2025-04-10:
Thanks for all the tests and feedback. Seems like flash size might be set somehow incorrectly there. I will take a look at it.

closed 2025-04-17 by espressif-bot

Dzarda7, 2025-04-17:
Hi @Fabchick, I managed to find the issue and fix it, appreciate the report! You can try it now with current master. New release will be done in matter of weeks, if you need it sooner, let me know.

Fabchick, 2025-05-02:
@Dzarda7 thank you very much for the quick help! I just tried it and it works. This project gets some thumbs up.