espressif/esp-serial-flasher #134: ESP32-WROOM-32UE-16MB bootloader issue

dflogeras, 2025-04-08:
Hello, I have a firmware on a STM32 micro which is using what appears to be parts of this library (I don't know the full history, but functions are named the same) to program the ESP32 module via UART with a custom firmware image.

I've noticed that about 5-10% of the time, the ESP32 enters bootload mode (via the GPIO0/EN pins sequencing), and responds with the appropriate string "ets Jul 29 2019 12:21:46\r\n\r\nrst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))\r\nwaiting for download\r\n", but then will not respond to any further programming sequences.

The next message sent from the STM32 to the ESP32, (when it works or not) is the byte sequence: 

"0xC0·0x00·0x08·0x24·0x00·0x00·0x00·0x00·0x00·0x07·0x07·0x12·0x20·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0x55·0xC0"

I've attached logic probe grabs of the same device working, and not working for reference.  Hopefully it is something silly I'm doing timing wise, but I cannot put my finger on it.  Is there a document which outlines the bootloader protocol?

Here is working:
1) ESP32 sends bootloader string
2) STM32 sends first programming byte sequence
3) After a timeout, STM32 sends sequence again
4) ESP32 responds and programming proceeds
[image]

And not working:
1) ESP32 sends bootloader string
2-11) STM32 sends first programming byte sequence, and times out after 10 attempts without an answer
[image]

Dzarda7, 2025-04-08:
Hello @dflogeras, the protocol is described [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/serial-protocol.html), I believe the issue you encountered is purely hardware-wise, but feel free to ask further questions. The `waiting for download` message is sent through UART indeed, but it is ignored, it is just informative, the whole protocol starts with the SYNC command, which is the next message you described. Thanks for pointing this out, I will try to add link for the protocol to the documentation.

dflogeras, 2025-04-08:
Hey @Dzarda7 thanks for your reply, and pointing me at the documentation.

Yup, I wasn't thinking that the 'waiting for download' had a side-effect, it just seemed to tell me that the ESP32 module is booted and running, and that it got my GPIO0 pin level strapped because it entered the boot-loader.  I can also confirm that the SYNC command our STM32 host is sending is exactly as the manual states (including the SLIP delimiters).

I've measured the outgoing SYNC message baud-rate (by averaging the time of sending the 32 0x55 bytes) and it seems to be about 1.2% slower than 115200.  Not sure if that is concerning.  I also measured the ESP32 115200 bit-rate from a working run and it also seems to be about the same speed (about 1.2% slower than ideal).  I have checked the errata of the device to see if anything was listed but came up empty.

Anything else you (or anyone else) can think of that I could probe at?  I don't think it is power, as the ESP32 boots up and seems to be happily printing the boot message about 200ms before I send the first SYNC.  Also, we aren't cutting the power supply, just the 'EN' pin on the module (the PUDR line on the chip).

Dzarda7, 2025-04-08:
So you say it works most of the time, but from time to time (5 to 10 %) it fails? Do you have something else connected to the UART pins of the ESP32? Is there for example USB-UART converter? I noticed an issue when the converter is present that it keeps the RX line of the target (in your case ESP32) high and the host (in your case STM32) is not able to pull enough current to force the line low enough. In this case 10 mA was not enough (20 mA should be, but might be the reason of this) and I forced the host to use 40 mA just to make sure (it was added [here](https://github.com/espressif/esp-serial-flasher/blob/bebb8d293c7fe4c60bb8053dede2b1aaa383ff80/port/esp32_port.c#L72)), not sure if it is your case, but might be worth to check it.

dflogeras, 2025-04-09:
Well to clarify, the update process seems to fail to SYNC every once in a while.  Normal app mode doesn't appear to suffer a start-up issue that I'm aware of.

Good suggestion on the signal levels, but I've now checked with a scope in analog-land and the pulses appear to be well formed and swing all the way from 0->3.3v.  The ESP32 is the only device connected on the UART.  I did however leave out a small detail, I'm not using the STM32 UART in this case, the system has an FPGA which synthesizes a UART, but electrically it is still a 1:1 relationship.

Another data point is that during our programming phase (and in the regular ESP32 app) we change to 230400 baud.  I would think our 1.2% error in baud rate (the same clock source generates it) would be felt more at the higher bit rate, but what I observe is that once the device is synchronized, the programming (at 230kBaud) works flawlessly.  We check a CRC before resetting.

Dzarda7, 2025-04-09:
I am not sure then, can you try it with the STM32 chip? Can you also try using the default baudrate for flashing to see if it also fails? There are multiple attempts done when one flashing command fails. You can also try to fiddle a bit with the SYNC command retry interval, maybe it fits into some clock pattern and it always fails also you can try to increase trials.

Dzarda7, 2025-04-30:
I am closing this for inactivity.

closed 2025-04-30 by Dzarda7