espressif/esp-serial-flasher #135: Does this project support running on ESP32-S3?

meibao-real, 2025-04-30:
Hello, I'm compiling and burning a project in ESP32-S3, but it can't connect to the target chip(esp32c3). Does this project support ESP32-S3?
The firmware code is from [esp-serial-flasher/examples/esp32_example], where the const loader_esp32_config_t config is configured as
```
const loader_esp32_config_t config = {
        .baud_rate = 115200,
        .uart_port = UART_NUM_1,
        .uart_rx_pin = U1RXD_GPIO_NUM,
        .uart_tx_pin = U1TXD_GPIO_NUM,
        .reset_trigger_pin = GPIO_NUM_0,
        .gpio0_trigger_pin = GPIO_NUM_2,
    };
```
The rest of the code is the same, but when I connect to the target chip, the monitor terminal shows a connection timeout
However, if I switch to ESP32, with exactly the same firmware code, it's connect to the target chip normally,
so Does this project support ESP32-S3?

Dzarda7, 2025-04-30:
Hi @meibao-real, it definitely should work with ESP32-S3. I would suspect wrong pins used. I would suggest to avoid using GPIO0 as it is strapping pin on ESP32-S3 (on ESP32 also, but there might be some differences). Can you try with different pinout? You can take inspiration from [esp32_spi_load_ram_example](https://github.com/espressif/esp-serial-flasher/tree/master/examples/esp32_spi_load_ram_example) as the pinout there is for ESP32-S3.

Dzarda7, 2025-05-02:
I tried with following pinout and it works, so either you have a HW issue or wrong pins used:
```
    const loader_esp32_config_t config = {
        .baud_rate = 115200,
        .uart_port = UART_NUM_1,
        .uart_rx_pin = GPIO_NUM_7,
        .uart_tx_pin = GPIO_NUM_6,
        .reset_trigger_pin = GPIO_NUM_5,
        .gpio0_trigger_pin = GPIO_NUM_13,
    };
```

meibao-real, 2025-05-03:
> I tried with following pinout and it works, so either you have a HW issue or wrong pins used:
>
> ```
>     const loader_esp32_config_t config = {
>         .baud_rate = 115200,
>         .uart_port = [...]
Hi [Dzarda7](https://github.com/Dzarda7), it works
very thanks

[image]

closed 2025-05-03 by meibao-real