espressif/esp-serial-flasher #136: esp32s3 USB flashing

dzurikmiroslav, 2025-05-01:
Hi, I developing esp32s3 project and I want connect HMI LVGL display also with esp32s3 (Crowpanel, VIEWE, gen4-ESP32 currently I don't know which one).
Would be nice to have only USB connector to HMI display, between them communication will be serial via AT commands. But sometimes host esp32s3 need to flash new firmware to HMI esp32s3.

My first try was connect secondary esp32s3 via USB to UART connector on devkit (CP2102), but controlling signals DTR/RTS not working, when I implement custom port on `usb_host_cp210x_vcp` begin working, also must change DTR/RTS bootloader sequence as has esptool.py https://github.com/espressif/esptool/blob/e132f6f761610c954f67108d1f5f0bb0724aa9f2/esptool/reset.py#L114
There is little disadvantage that this project not support user defined ports in esp-idf build system...

`esp32_usb_cdc_acm_port` work nice when target esp32s3 has enabled `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG` and is connected to USB OTG.

Does `esp32_usb_cdc_acm_port` support bootloader DTR/RTS sequence on any USB to UART converter?

Or is a better approach, to leave enabled  `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG` on target esp32s3 and connect via USB OTG? I am not sure if is OK to leave enabled JTAG in production?

Thanks in advance for all the recommendations :)

Dzarda7, 2025-05-02:
Hello @dzurikmiroslav, the choice between USB-serial-JTAG and USB-UART converted is up to you, but I can tell you that USB-serial-JTAG would be probably faster (in esptool it is ofter around twice as fast as UART transfer). I believe it is not that big of a deal for occasional FW update, just so you know.

Second thing is that I believe `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG` has no effect here. It just outputs the log of the app to the USB instead of UART. `esp32_usb_cdc_acm_port` does not depend on this option enabled. JTAG is enabled in both cases anyway I believe. It can only be disabled by burning [appropriate fuses](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/jtag-debugging/configure-other-jtag.html).

Regarding the UART converters support, the support for CP21XX and CH340 will be hopefully soon rewritten from C++ to C and can be added to this project (just waiting for the guys to merge [this](https://github.com/espressif/esp-usb/commits/feat/vcp_c_api/)). Then I need to figure out some nice way to choose which one to use. You probably have some idea as you mentioned:

> There is little disadvantage that this project not support user defined ports in esp-idf build system...

Can you please elaborate on this a bit more? I can keep it in mind while implementing the feature.

Last think, I need to warn you that I just recently found that there is an issue with `esp32_usb_cdc_acm_port`. If you run it for around 1,5 hour, it fails with TX timeout. For the 1,5 period, you can do flashing as many times as you want. It it most probably not an issue with this library, might be in [usb_host_cdc_acm](https://components.espressif.com/components/espressif/usb_host_cdc_acm), but we are not sure yet. It also does not have to be issue when `usb_host_cp210x_vcp` will be used (I need to test it after the merge). This might not bother you as you will just communicate with the target, then init esp-serial-flasher, flash, deinit esp-serial-flasher and continue with communication. I just recommend you to try this if it is an issue for you. Sorry for this, hopefully it will be resolved soon.

dzurikmiroslav, 2025-05-02:
Hello, thanks for reply

Standard communication between ESPs will be AT commands, but for flashing I need enter bootloader mode.

For scenario USB-OTG I think there is two options:

1. In target ESP32 when I use stdout/stdin for AT commands and enable `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG`. When running my firmware, there is available USB JTAG dev (VID=0x303a PID=0x1001) and I can communicate with AT commands. USB JTAG can communicate with esp32_usb_cdc_acm_port and work enter to bootloader mode.

2. Run AT commands on USB CDC (`CONFIG_ESP_CONSOLE_USB_CDC` or TinyUSB), then I must implement custom AT command to switch target ESP to bootloader mode and then start esp32_usb_cdc_acm_port flashing.

USB-OTG also has the disadvantage that there are options to brick target ESP (when JTAG is not start with firmware and need to manualy enter bootloader mode).
With USB-to-UART converter has full control of target ESP, anytime can be switched to bootloader mode via DTR/RTS signals.

> There is little disadvantage that this project not support user defined ports in esp-idf build system...

It would be enough to add the `SERIAL_FLASHER_INTERFACE` option like `SERIAL_FLASHER_INTERFACE_USER` and then, user could define own implementation of port methods `loader_port_read`, `loader_port_write`, `loader_port_enter_bootloader`... for example based on usb_host_vcp library.

But if this project will supports standard USB to UART converters, this will not be necessary. I look forward to it :)

Dzarda7, 2025-05-05:
Regarding the "brick", it is not entirely true, the issue is that in some cases, watchdog can get triggered, port disappear and flashing fails. This can be solved by turning the watchdog off and I am thinking about implementing this, but there are other issues with this implementation. Also, when the target is put into bootloader mode for example by BOOT and RESET buttons, USB-serial-JTAG flashes correctly, but the reset afterwards does not trigger reading of strapping pins, so previous state is used and it boots again into bootloader mode, solution is to reset the device by BOOT button or power off and on. Anyway, as you said. USB-UART converters are potentially less hassle, but I cannot guarantee a timeline when it will be supported.

Regarding the custom port. You actually can do that and it is pretty simple. It is even mentioned in the main README in [Supporting a New Host Target](https://github.com/espressif/esp-serial-flasher?tab=readme-ov-file#supporting-a-new-host-target) section. You can take inspiration from Raspberry pi pico support, which was recently [added](https://github.com/espressif/esp-serial-flasher/commit/c72680c6790843ddf49274a3ba8442fec5eca0ff).

dzurikmiroslav, 2025-05-05:
Yes, I did it this way. I must write own `CMakeLists.txt` for this component, that means the component cant be installed via IDF Component Manager (`idf_component.yml`).

I don't think it's possible to overwrite managed component `CMakeLists.txt`?

Dzarda7, 2025-05-05:
I believe it might be possible when you bundle it with the project and specify the path in `idf_component.yml`, but you can also move it into components folder.

dzurikmiroslav, 2025-05-05:
Thanks for the recommendation, I'll try this approach.
I'm definitely looking forward to supporting USB 2 UART converters in the future :)

closed 2025-05-05 by dzurikmiroslav