espressif/esp-serial-flasher #137: Compiler warnings

ZiGaMi, 2025-05-02:
### Port

STM32 using CubeIDE

### Target chip

ESP32C3

### Hardware Configuration

Using custom PCB, standard UART connection between STM and ESP

### Log output

```plain
Compiler warnings does not procude log output
```

### More Information

Compiling with following warnings enable:

[image]

Results in bunch of compiler warnings (with enabled "tread warnings as error") it results in bunch of error:

[image]

Can you please consider addressing those warnings 🙏

Dzarda7, 2025-05-02:
Hi @ZiGaMi, I believe that the issue is that STM32CubeIDE does not take [CMakeLists.txt](https://github.com/espressif/esp-serial-flasher/blob/master/CMakeLists.txt) of the esp-serial-flasher library into an account. The file ensures that the correct files are built. For example `loader_port_write` function is defined differently for SDIO and for UART ports. I will try to take a look at this, but I am not that familiar with importing CMake projects into STM32CubeIDE.

ZiGaMi, 2025-05-05:
Thank you @Dzarda7 for your answer!

OK, I see now that you rely on build system to exclude files that are irelevant to the given configuration. That makes direct coupling to CMake so for other build systems end user shall deal with it. If it makes sense I'd recomend to #ifdef/#endif complete (or part of the) file instead, to doit in code and that should put CMake dependecy out of the equation.

Another things are implicit conversion warnings. I believe in zero warning code thats why I'm asking for that clean up :)

Can we expect a zero warning library soon? 

Thank you again for your time!

Dzarda7, 2025-05-06:
When I compile the example with [__build_set_default_build_specifications](https://github.com/espressif/esp-idf/blob/aaebc374676621980878789c49d239232ea714c5/tools/cmake/build.cmake#L119), it is without errors and warnings, so I believe the issue is that STM32cubeIDE ignores the CMakeLists.txt. It might be possible to use it there, but I did not have time to test it. I think that do so many #ifdef/#endif would be quite messy solution and not sure if I would be able to get rid of CMake entirely.

Dzarda7, 2025-05-08:
Thank you @ZiGaMi for bringing this out, I did not try the example with STM32CubeIDE before, but I thought about some improvements now and it can be definitely better. The IDE can import CMake projects, so that should not be an issue. That should also take care of the warnings you mentioned. I now have some higher priority stuff unfortunately, so I will take a look at it later if it is not an issue for you.