espressif/esp-serial-flasher #138: Purpose of "loader_port_change_transmission_rate"

ZiGaMi, 2025-05-05:
Why is the ***loader_port_change_transmission_rate*** function even part of the "port" layer if it's not being used by the library? 

Maybe it would make sense if higher baudrate will be option in *connect_config*, something like that:
```C
/**
 * @brief Connection arguments
 */
typedef struct {
    uint32_t sync_timeout;  /*!< Maximum time to wait for response from serial interface. */
    int32_t trials;         /*!< Number of trials to connect to target. If greater than 1,
                               100 millisecond delay is inserted after each try. */
   
  // ADDED FOLLOWING ARGS
   bool switch_baudrate;        // Swtich baudrate to "high_baudrate" using "loader_port_change_transmission_rate" fucn
   uint32_t high_baudrate;
} esp_loader_connect_args_t;

#define ESP_LOADER_CONNECT_DEFAULT() { \
  .sync_timeout = 100, \
  .trials = 10, \
  .switch_baudrate = true, \
  .high_baudrate = 460800, \
}

// And then when calling it makes everything for you :)
const esp_loader_connect_args_t connect_config = ESP_LOADER_CONNECT_DEFAULT();
const esp_loader_error_t err = esp_loader_connect((esp_loader_connect_args_t*) &connect_config );

```  

What do you guys think about that? Hope that makes sense :)

Dzarda7, 2025-05-05:
As mentioned in the main [README](https://github.com/espressif/esp-serial-flasher?tab=readme-ov-file#supporting-a-new-host-target), the function is just for convenience and is not used by the library. Same with `esp_loader_reset_target`, you do not need to use that, you can just toggle the reset pin.

Regarding the change, thanks for the suggestion. It might make sense for flashing only purpose, but let's say for reading target info, burn fuses (as a planned feature) and other commands that does not exchange big amounts of data, it is unnecessary overhead to change baudrate. Like this you can call it anytime you want, similarly as esptool which changes baudrate only before flashing. Also this would not be applicable for other ports which either change baudrate at the beginning of the transaction (like sdio) or cannot change the baudrate at all (like `usb_cdc_acm` port). Last but not least, it would be breaking change which can be done only when v2 is released.

So overall thanks for the suggestion, but I do not think it is a good idea.

Dzarda7, 2025-05-06:
Okay, I thought you want to remove `esp_loader_change_transmission_rate` from public API. Like this it does not introduce breaking change, but I am still not big fan of it as `ESP_LOADER_CONNECT_DEFAULT` is used for all ports and now there will be options that does not apply for other ports. Also it makes it a bit harder for user to immediately see, what is going on I believe, which I am also not big fan of. I will think about it, but I will probably not implement this, but thanks for the suggestion. It might make sense to have one function to change baudrate for host and target.

closed 2025-05-12 by dobairoland