espressif/esp-serial-flasher #139: ESP flash fails and reverts to version 0.0.0.0, UART Flashing becomes unusable until recovery

rlux-bluezone, 2025-05-23:
### Port

STM32

### Target chip

ESP32-C6

### Hardware Configuration

Using custom PCB, standard UART connection between STM and ESP

### Log output

```plain
DEBUG: Error: DEBUG: FLASH_WRITE_ERRDEBUG: <\n>
DEBUG: Error: DEBUG: FLASH_WRITE_ERRDEBUG: <\n>
DEBUG: Error: DEBUG: FLASH_WRITE_ERRDEBUG: <\n>
DEBUG: Packet could not be written! Error INVALID RESPONSE.<\n>
```

### More Information

I'm updating the ESP, and it's working most of the time! But sometimes I get the error listed above(ESP_LOADER_ERROR_INVALID_RESPONSE). When I get the error, the flash fails, and the ESP now shows version 0.0.0.0. From this point on, flashing the ESP via UART doesn't work.

I have the option to flash it using the ESP tool. After flashing with it, the ESP becomes usable again, and I can flash it via UART until the error occurs again.

It's ok that the flash sometimes fails, but how can I get the ESP back into an updateable state via the serial flasher?

Dzarda7, 2025-05-26:
For me it seems like some hardware issue for now. You do not need to care about IO8 I believe for C6. It can be kept HIGH always. The pin handling is done by library, so you just need to specify the IO0 and RESET pin in the [structure](https://github.com/espressif/esp-serial-flasher/blob/master/examples/stm32_example/Core/Src/main.c#L107). Unfortunately the IO0 in the structure represents the BOOT pin which is IO9 in case of ESP32C6. This confusion will be solved in 2.0 release, so sorry for that. What I want to say is that you do not need to handle the reset of the target by yourself, it is handled by the library. So this can be the issue, but otherwise I am not entirely sure what might cause this. You can try to send the schematic and init code and I can take a look.

Dzarda7, 2025-05-27:
> You mean when I connect via the USB Serial when the problem occurs i should get

Yes, that is what you should get when esp-serial-flasher tries to get the ESP32C6 into bootloader mode, this means it is successful. But as I said, there is no way incorrect flashing causes imposibility to get target to download mode by toggling BOOT and RESET pin. If you will be able to reproduce it, I can help you more. Like this I don't think I can do much. I will keep the issue open for a while, so feel free to keep trying and asking questions. If nothing new comes up, I will have to consider this HW issue.

Dzarda7, 2025-05-29:
@rlux-bluezone We have new AI issue bot that and there was a problem with it. It responded now so sorry for confusion, but it might be helpful.

Dzarda7, 2025-06-20:
Hi @rlux-bluezone, did you find something more? Were you able to reproduce the issue? Otherwise I would have to close it, because it seems like hardware issue with the pins controlling RST and BOOT.

Dzarda7, 2025-08-22:
Hello @rlux-bluezone, can you please verify, that the same blocks written consecutively are indeed completely the same? The 01 08 error indicates that received data are incorrect according to checksum (which is sent with each packet) as you can see in [commands](https://docs.espressif.com/projects/esptool/en/latest/esp32h2/advanced-topics/serial-protocol.html#commands). Either checksum or data is incorrectly received. Then invalid response is sent, esp-serial-flasher tries it again. Your observation is correct, sequence number is incremented in this case too. It is correct behavior, because ROM code increments it even though data has incorrect checksum. What's more, sequence number has no influence on the address written, it is automatically incremented. Interesting is that address should not be incremented if incorrect data is received or written to flash. Seems like the data is written correctly, but response is wrong, I really cannot say how this can happen. Are you able to flash the chip again afterwards? Did you solve this issue? You can try to enable debug trace and send that to me.