espressif/esp-serial-flasher #151: Stuck at Erasing flash when run on Linux platform (OpenWRT)

zhengruiw02, 2025-08-15:
### Port

Raspberry-pi port (modified for OpenWRT by remove pigpio_LIB pigpio, and use fp control on /sys/class/gpio/ )

### Target chip

ESP32-C6

### Hardware Configuration

MT7688 GPIO1 to ESP32-C6 EN (inverted logic)
MT7688 GPIO3 to ESP32-C6 IO9.
MT7688 UART1_TX to ESP32-C6 RX
MT7688 UART1_RX to ESP32-C6 TX.

### Log output

```plain
Export pin 513
Set direction pin 513 as out
Export pin 515
Set direction pin 515 as out
Write pin 515 as 0
Write pin 513 as 1
Write pin 513 as 0
Write pin 515 as 1

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 
--- READ ---
45 53 50 2d 52 4f 4d 3a 65 73 70 33 32 63 36 2d 32 30 32 32 30 39 31 39 0d 0a 42 75 69 6c 64 3a 53 65 70 20 31 39 20 32 30 32 32 0d 0a 72 73 74 3a 30 78 31 20 28 50 4f 57 45 52 4f 4e 29 2c 62 6f 6f 74 3a 30 78 34 20 28 44 4f 57 4e 4c 4f 41 44 28 55 53 42 2f 55 41 52 54 30 2f 53 44 49 4f 5f 46 45 49 5f 46 45 4f 29 29 0d 0a 77 61 69 74 69 6e 67 20 66 6f 72 20 64 6f 77 6e 6c 6f 61 64 0d 0a 
--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 
--- READ ---
c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 c0 01 08 04 00 07 07 12 20 00 00 00 00 c0 
--- WRITE ---
c0 00 14 00 00 00 00 00 00 c0 
--- READ ---
c0 01 14 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0c 0d 00 00 00 00 00 00 00 00 00 00 00 c0 
--- WRITE ---
c0 00 0d 08 00 00 00 00 00 00 00 00 00 00 00 00 00 c0 
--- READ ---
c0 01 0d 04 00 00 00 00 00 00 00 00 00 c0 Connected to target

--- WRITE ---
c0 00 0f 08 00 00 00 00 00 00 08 07 00 00 00 00 00 c0 
--- READ ---
c0 01 0f 04 00 00 00 00 00 00 00 00 00 c0 Transmission rate changed.
Loading bootloader...
Erasing flash (this may take a while)...

--- WRITE ---
c0 00 0a 04 00 00 00 00 00 18 30 00 60 c0 
--- READ ---
```

### More Information

I checked on ESP32-C6 UART print, and found it can be in UART download mode when I ran esp-serial-flasher. Print log as below.
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x4 (DOWNLOAD(USB/UART0/SDIO_FEI_FEO))
waiting for download
```

Dzarda7, 2025-08-15:
Hello @zhengruiw02, so what is the issue, you do not get response on `c0 00 0a 04 00 00 00 00 00 18 30 00 60 c0` can you please elaborate a bit more?

zhengruiw02, 2025-08-15:
Hi @Dzarda7 , I solve it after I change `HIGHER_BAUD_RATE` to 115200. Now I can successfully flash the target.

Close this issue since solved.

Dzarda7, 2025-08-15:
But we can try to solve it anyway 😄. If you are interested, I believe you have have forgot to change baudrate of the rpi when after `Transmission rate changed.`. Feel free to ignore as it works it for you, but I believe this is the issue.