espressif/esp-serial-flasher #152: Add stub support for ESP32S3 - waiting for esp-flasher-stub

b1t916, 2025-08-19:
### Port

ESP32S3

### Target chip

ESP32S3

### Hardware Configuration

Custom Board use ESP32-S3-WROOM-1U as host
Target Chip: ESP32-S3-WROOM-1U
Baudrate: 115200
esp-serial-flasher version: 1.9.0

### Log output

```plain
ESP_LOADER_ERROR_TIMEOUT
```

### More Information

My board use ESP32-S3-WROOM-1U as host, when i flash binaries to another ESP32-S3-WROOM-1U  use stub mode, badurate 115200, log output error: ESP_LOADER_ERROR_TIMEOUT.
This happens at invoke "esp_loader_connect_with_stub" function.
But i can flash ESP32-S3-WROOM-1U without stub mode.(eg. use esp_loader_connect)

b1t916, 2025-08-25:
> Hello [@b1t916](https://github.com/b1t916), can you provide more info please? I am unable to reproduce the issue. Maybe code, HW connection.

It seems there's a problem when switching from the Rom Loder method to the stub method. If there were an issue with the hardware connection or the underlying code, one would expect the Rom Loder method to fail as well. However, since the Rom Loder is functioning correctly while the stub is not, the issue appears to be directly related to the change from `esp_loader_connect` to `esp_loader_connect_with_stub`.

Dzarda7, 2025-08-28:
Happy you solved it.

closed 2025-09-01 by dobairoland