espressif/esp-serial-flasher #153: Add support for ESP32-C5 flashing over SDIO interface

dmartauz, 2025-08-20:
### Is your feature request related to a problem?

ESP32-C5 has potential to become preferred WiFi co-processor due to 5GHz band support.

### Describe the solution you'd like

Implementation for both 1-bit (preferred) and 4-bit SDIO would be appreciated.
Please clarify recommended SDIO pin and ESP32-C5 strapping pin configuration in the documentation.

dmartauz, 2025-09-19:
@Dzarda7 is there any update? Could you please recommend proper SDIO pin and ESP32-C5 strapping pin configuration? We would like to put our board into manufacturing next week.

dmartauz, 2025-09-22:
MTMS pin (IO2) must be set properly as well.

[image]

Dzarda7, 2025-09-22:
BTW I believe at least modules will come with pre-burned `XTAL_48M_SEL_MODE` as devkits does.

dmartauz, 2025-09-22:
We included pull-up/down resistor footprints for MTMS just to be sure.

dmartauz, 2025-10-17:
@Dzarda7 are you sure regarding C5 production state? We have several samples of C5-WROOM modules and I was thinking those are MP version. We did not assemble any of them on the PCB yet as we focus on the MIPI interface of P4 right now. I would appreciate if the C5 could be supported in a month or two. Is it feasible?

Rokachy, 2025-11-17:
Thank you!!
Please notify us when it ready.

Dzarda7, 2025-12-04:
Also beware that SDIO is really susceptible to signál integrity problems, so try to use short traces and match trace length width.

dmartauz, 2026-05-11:
@Dzarda7 since Joint Download Boot 1 mode of ESP32-C5 outputs `DOWNLOAD(UART0/SDIO_FEI_FEO)` I believe the correct edge setting is the first option, please confirm:
[image]
I managed to run `esp32_get_target_info_example` modified for SDIO transport and RESET/BOOT pins driven by `esp_io_expander_gpio_wrapper` and I am getting flash size and MAC info. I have MTDI and GPIO25 floating, however I think there is no internal pull-down so they probably should be pulled externally to avoid floating issues.