espressif/esp-serial-flasher #155: Upgrading from 1.8 to 1.9 resulted in the inability to burn 8266

vvvyky, 2025-09-21:
### Port

ESP8266

### Target chip

ESP8266

### Hardware Configuration

Upgrading from 1.8 to 1.9 resulted in the inability to burn 8266, but my program has not changed. May I ask what changes have occurred in 1.9 that affect the burning of 8266

### Log output

```plain
V1.8
Last burning template: ESP8266-UART_iO0X0.bin, 0x0512000
Start burning template: undefined
Data: ESP8266-UART_iO-0X0.bin, 0x0512000
Burn firmware: ESP8266-UART_iO0X0.bin Address: 0 Speed: 512000
Serial port not initialized, initialize serial port ..
Initialize serial port ..115200
Initialize serial port ..115200
Serial port connection is in progress ..
Connection failed, please check if there is an error in the connection!
Connection failed, flip the serial port and retry!
Initialize serial port ..115200
Serial port connection is in progress ..
ESP8266 successfully connected!
Initialize serial port ..512000
Serial port connection is in progress ..
ESP8266 successfully connected!
Read chip details!
Chip model: ESP8266
Flash memory: 4MB
MAC:Unknown MAC
Target chip: Unrecognized
Ecological version number: 1073683720
Safe Startup: Enable
Flash Encryption: Enable
Safe download mode: enabled
Secure Startup Key 0 Cancel: Enable
Secure Startup Key 1 Cancel: Enable
Secure Startup Key 2 Cancel: Enable
JTAG access: permanently disabled
USB access: disabled
Data caching in UART download mode: disabled
Instruction cache in UART download mode: disabled
ESP8266 successfully connected!
initialization. ..
Burning in progress ..
Last burning template: ESP8266-UART_iO0X0.bin, 0x0512000
ESP8266-UART_iO0X0.bin successfully burned!
Last burning template: ESP8266-UART_iO0X0.bin, 0x0512000
1 firmware burned successfully | 0 firmware burned failed!
Remaining burning times: 9969
Initialize serial port ..115200

V1.9
Start burning template: ESP8266-UART_iO-0X0
Data: ESP8266-UART_iO-0X0.bin, 0x0512000
Burn firmware: ESP8266-UART_iO0X0.bin Address: 0 Speed: 512000
Serial port not initialized, initialize serial port ..
Initialize serial port ..115200
Initialize serial port ..115200
Serial port connection is in progress ..
Connection failed, please check if there is an error in the connection!
Connection failed, flip the serial port and retry!
Initialize serial port ..115200
Serial port connection is in progress ..
Connection failed, please check if there is an error in the connection!
ESP8266-UART_iO0X0.bin burning failed!
0 firmware burned successfully | 1 firmware burned failed!
Initialize serial port ..115200
Start burning template: ESP8266-UART_iO-0X0
Data: ESP8266-UART_iO-0X0.bin, 0x0512000
Last burning template: ESP8266-UART_iO0X0.bin, 0x0512000
Burn firmware: ESP8266-UART_iO0X0.bin Address: 0 Speed: 512000
Initialize serial port ..115200
Serial port connection is in progress ..
Connection failed, please check if there is an error in the connection!
Connection failed, flip the serial port and retry!
Initialize serial port ..115200
Serial port connection is in progress ..
Connection failed, please check if there is an error in the connection!
ESP8266-UART_iO0X0.bin burning failed!
0 firmware burned successfully | 1 firmware burned failed!
Initialize serial port ..115200
```

vvvyky, 2025-09-21:
ESP-srial-flasher1.9 built on ESP32 cannot burn 8266, while ESP-srial-flasher1.9 built on C3 can burn 8266 normally without any solution?

Dzarda7, 2025-09-22:
Hello @vvvyky, thanks for the report. I tried flashing ESP8266 from ESP32 with v1.8.0, v1.9.0 and master branch, all worked correctly. If you saying ESP32 does not work, but ESP32-C3 works, it might be some issue with ESP-IDF (which I believe is highly unlikely as UART driver is well tested), but can you try different ESP-IDF version please? I tried [esp32_example](https://github.com/espressif/esp-serial-flasher/tree/master/examples/esp32_example).

Dzarda7, 2025-09-26:
So you saying issue was only in ESP-IDF v5.5.0 and v5.5.1 release fixed the issue? I did not see anything related to UART in ESP-IDF v5.5.1 release notes. But I believe it is safe to consider this fixed right? Will take a look, but if you agree, we can close this. 

Btw out of curiosity, you use some custom Arduino esp-serial-flasher port? I experimented with it a bit too and thought if it makes sense to do one.

Dzarda7, 2025-10-06:
I am closing this for inactivity, I was unable to reproduce the issue.